microtherion/ScratchMonkey #2: MCU Target ATmega2560 support

Georges760, 2016-02-26:
Hi,
I am currently working onto a custom ATmega2560 based board. On one of my prototype, when programming the fuse for the first time, a short-cut should have wrongly programmed the SPIEN... So I need to use HVPP now to restore the fuses. I just came accros your great project and looking into documentation, I see "would need to find board w/o external clock to test HVPP" about MCU Target ATmega2560 support.
I understand HVPP should have a external clock free target MCU to success. My custom board have an external oscillator, but I can remove it the time to fix fuses.
Would you mind explain the theorical wiring diagram, and maybe i can do the testing for you.

Thanks

Georges760, 2016-02-26:
Looking into Datasheet, I found the need for Pa0, all PB and all PD pins, together with XTAL1. Look similar to ATmega1284 that you already trested.

My custom board does not have PA0 and PD pins routed of MCU, so I will not be able to help you.

But you could try anyway using a Arduino Mega board with a removed oscillator and a soldered wire on XTAL1...

Thanks again for this great project.

microtherion, 2016-02-26:
Sorry that HVPP won't work for you. Have you tried ScratchMonkey in ISP mode? It uses some techniques (software SPI) to work around problematic fuse settings (i.e. it can't fix SPIEN, but it can deal with accidentally specifying the internal 128kHz oscillator).