microtherion/ScratchMonkey #3: Arduino IDE 1.6.7 Won't Recognize Scratch Monkey Hardware Definition?

paulrpotts, 2016-11-17:
I am running the Arduino/Genuino 1.6.7 IDE on Windows 7. I have an Uno R3 and can build and program the Scratch Monkey sketch to this board.

However, I can't get any of the Scratch Monkey listings to show up in the Tools menu.

I see "WARNING: Error loading hardware folder C:\Users\ppotts\Documents\Arduino\hardware\scratchmonkey
  No valid hardware definitions found in folder scratchmonkey."

The programmers.txt file is copied directly from the ScratchMonkey distribution folder.

Any ideas?

I have the hardware all wired up to try using HVPP to program an ATtiny 641 including a 12V supply following the switching schematic, but can't test it.

Thanks!

microtherion, 2016-11-17:
The Arduino 1.6 IDE has a rather different file layout compared to earlier versions, and to my shame, I still haven't made the switch, so I don't know what changes exactly are necessary.

paulrpotts, 2016-11-17:
Oh! OK. So I won't expect it to work at all for the time being. I'll see if I can figure out some other way to do it for the time being. Thanks for your reply.

pgs21, 2017-02-02:
Got it working in arduino 1.8.1

Documents/
    Arduino/
       hardware/
           scratchmonkey/
               avr/
                   boards.txt
                   programmers.txt

boards.txt is a empty file

paulrpotts, 2017-02-02:
Glad to hear it! I wound up buying an Atmel Dragon for some chips that needed the high voltage programming treatment and that worked (what a mess of wires, though!)

DCEM, 2017-03-03:
Hi,

I got it working thourgh the board manager ... I would prepare a pull request if you (microtherion) want to continiue supporting ScratchMonkey for Arduino IDE 1.6 plus... 
(I need to know because otherwise i would let the links in the json file point to my repository ...)

Also while trying to figure out how to realise the board manager stuff I noticed that if you compile ScratMonkey on Arduino 1.8.1 it does not work. The version compiled with arduino 1.0.6 works just fine ...

Trying to sort this out at the moment...

I think it should be possible to add a patched avr-dude version, so that it is basically a one click solution to get the hardware running with the scratchmonkey protocol ...

Best regards!

microtherion, 2017-03-03:
Would love to get a PR for your changes. I recently acquired a Due compatible board so I could start testing on a non-AVR platform.

What protocol did not work?

DCEM, 2017-03-03:
Hi,

Great, wasn't sure that you were still active on this project. I will prepare the PR shortly

I tried it with ScratchMonkey (STK500 Mode) -> ISP 
I rememberd that the problem might be related to the changes in compiler optimisation from Arduino 1.0.6 to 1.6.x+ 

I'm gussing that this screws up your timings....

See:
http://www.instructables.com/id/Arduino-IDE-16x-compiler-optimisations-faster-code/

microtherion, 2017-03-03:
That makes sense, to some extent, although hardware ISP is pretty much the least handwritten protocol in the suite. Maybe some setup times which were automatically assured with a slow compiler and thus never explicitly implemented?

ingegno, 2018-03-30:
Anything I can try to have this working? Would love to stop using 1.0.5 ...