microtherion/ScratchMonkey #6: HVPP for mega32l not working

ameeuw, 2017-09-08:
This is a really great project and I hope that it can save me from having to buy a separate HV capable programmer...

My problem is that I burnt the fuses of an mega32l chip in a charger (iMax B6AC) wrong and now it is unresponsive to the ISP interface.

I have connected the chip (with a mess of wires - see attached image) to my ScratchMonkey based on an Arduino Uno.
[image]

Now I am trying to first read out the fuses using:

`avrdude -p m32 -c stk500pp -P /dev/cu.usbmodem1421 -U lfuse:r:low_fuse_val.hex:h -U hfuse:r:high_fuse_val.hex:h`

but all I get over time is:
`avrdude: stk500v2_ReceiveMessage(): timeout`

however when I unplug the CTRL0 wire (which is shared with the serial RX) there seems to be a sync and avrdude communicates with SM:

`avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x000000 (retrying)`

Of course the PP is not working anymore since I removed the CTRL0/BS2 wire... as I am not 100% familiar with avrdude I am not sure whether I am making a mistake with the command line or whether I miswired something on scratchmonkey (checked it multiple times).

I am very happy to be pointed into a direction on how to resolve these issues.. would be nice to revive my dead charger and flash some custom firmware on it.

Best regards,
Arne

ameeuw, 2017-09-08:
I went over the problem today again and desoldered a capacitor on the CTRL0/BS2 line and now SM works.
As I am getting no readout that makes sense:
`Arnes-MacBook-Pro:backup arne$ avrdude -p m32 -c stk500pp -P /dev/cu.usbmodem1421 -F

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x000100
avrdude: Expected signature for ATmega32 is 1E 95 02

avrdude: safemode: Fuses OK (E:FF, H:0E, L:00)

avrdude done.  Thank you.
`

I am assuming now, that there are probably other capacitors on the control and data lines that are coupling and therefore dampening the signals going in and out.

My question here is now: Is there a way to rescue this device without desoldering and resoldering the mega32 processor?

I am still thankful for any answers - thank you.

microtherion, 2017-09-08:
Unfortunately, HVPP can be rather finicky to get right because of the large number of signals involved. Doing this with a chip soldered to a board is even trickier. One common complication is that if the board uses a crystal, that will most likely be connected to XTAL1 and almost certainly interfere with the HVPP signal connected there.