microtherion/ScratchMonkey #7: mega328p got error when buring bootloader

raffeale, 2017-12-02:
hi , i use HVPP to repair fuse with a Arduino UNO ,and got success message.

and then i burning bootloader with ScratchMonkey(STK500 MODE),
i got following message. i don't know how to fix it.

raffeale@creator:~/arduino-1.0.6$ /home/raffeale/arduino-1.0.6/hardware/tools/avrdude -C/home/raffeale/arduino-1.0.6/hardware/tools/avrdude.conf -v -v -v -v -patmega328p -cstk500v2 -P/dev/ttyUSB0 -e -Ulock:w:0x3F:m -Uefuse:w:0x05:m -Uhfuse:w:0xde:m -Ulfuse:w:0xff:m  -F

avrdude: Version 5.11, compiled on Sep  9 2011 at 16:00:41
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/home/raffeale/arduino-1.0.6/hardware/tools/avrdude.conf"
         User configuration file is "/home/raffeale/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : stk500v2
avrdude: Send: . [1b] . [01] . [00] . [01] . [0e] . [01] . [14] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: Send: . [1b] . [01] . [00] . [01] . [0e] . [01] . [14] 
avrdude: Recv: . [1b] 
avrdude: Recv: . [01] 
avrdude: Recv: . [00] 
avrdude: Recv: . [10] 
avrdude: Recv: . [0e] 
avrdude: Recv: . [01] 
avrdude: Recv: . [00] 
avrdude: Recv: . [0d] 
avrdude: Recv: S [53] 
avrdude: Recv: C [43] 
avrdude: Recv: R [52] 
avrdude: Recv: A [41] 
avrdude: Recv: T [54] 
avrdude: Recv: C [43] 
avrdude: Recv: H [48] 
avrdude: Recv: M [4d] 
avrdude: Recv: O [4f] 
avrdude: Recv: N [4e] 
avrdude: Recv: K [4b] 
avrdude: Recv: E [45] 
avrdude: Recv: Y [59] 
avrdude: Recv: O [4f] 
avrdude: stk500v2_getsync(): got response from unknown programmer SCRATCHMONKEY, assuming STK500
avrdude: stk500v2_getsync(): found STK500 programmer
         AVR Part                      : ATMEGA328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
[... 136 lines cut ...]
avrdude: Recv: . [1b] 
avrdude: Recv: . [0c] 
avrdude: Recv: . [00] 
avrdude: Recv: . [02] 
avrdude: Recv: . [0e] 
avrdude: Recv: . [10] 
avrdude: Recv: . [c0] 
avrdude: Recv: . [cb] 
avrdude: stk500v2_command(): command failed
avrdude: initialization failed, rc=-1
avrdude: AVR device initialized and ready to accept instructions
avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
avrdude: Expected signature for ATMEGA328P is 1E 95 0F
avrdude: Send: . [1b] . [0d] . [00] . [03] . [0e] . [11] . [01] . [01] . [0a] 
avrdude: Recv: . [1b] 
avrdude: Recv: . [0d] 
avrdude: Recv: . [00] 
avrdude: Recv: . [02] 
avrdude: Recv: . [0e] 
avrdude: Recv: . [11] 
avrdude: Recv: . [00] 
avrdude: Recv: . [0b] 

avrdude done.  Thank you.

microtherion, 2017-12-02:
I assume you connected in ISP mode? Did you follow the connection diagram?

Depending on what you set the fuses to, you might need to apply the clock signal (Marked as `(XTAL)` in the diagram).

raffeale, 2017-12-02:
i connected all pin  based on  https://microtherion.github.io/ScratchMonkey/Connection.html
programmer is  Arduino UNO , target MCU Mega328P.
i connected XTAL1 PIN of target 328P to Arduino Uno 9 PIN.
i use arduino 1.0.6 IDE  to  compile and upload ScratchMonkey in Arduino uno.
then i choose programmer to ScratchMonkey(STK500 Mode) to burning bootloader , i got error message.
and then i try  arduino as ISP programmers to burning bootloader , i got error mesage too!

raffeale, 2017-12-02:
[image]
i connected all pin by this diagram  to repair fuse.

[image]

i am burning bootloader depends on  this diagram.  i choose  ScratchMonkey(STK500 Mode)  for programmer

onre, 2018-03-02:
FWIW, I'm hitting the same bug. avrdude version 6.3 as supplied with latest Arduino IDE on Windows, if that matters. ISP mode, trying to reset a fuse, have the XTAL out connected to XTAL2 on target as it is in the "watch crystal" mode.