microtherion/ScratchMonkey #12: ArduinoMicro does'nt work

takachinss1001jp, 2020-12-08:
Dear Sir,
 Will you Please tell me,What Happed on Arduino Micro?
My The following circumstances are folloing;
 I tried to Assemble ScratchMonky HVPP and TPI Writer with Arduino Micro according User Manual Target Connection Diagram.
 However It does'nt work.

 So Analized what happed on sketch, The Portmapping seemed to wrong.  Because folllowing code seemed to unopen the Port.
typedef SMoHWIF_Port_Dual<SMoHWIF_PORT_D, 0x0C, 0,   // Only Open CTRL2 & 3
                          SMoHWIF_PORT_F, 0xF1, 0>                          //                   CTRL0, CTRL4,CTRL5,CTRL6 & CTRL7
Therefore I think CTRL1,DATA0,DATA1,DATA2,DATA4 are  Disabled, And Seemed Undefined DATA5 on  PORTE DATA3 on PORTC is not Defined.

I'm look forward your Kindness Cooperation.

Kazuhiro Haitaka.

microtherion, 2020-12-11:
Thanks for your question! I'm afraid I'm very rusty on this code, so please give me a few days to think about it and remind me if you haven't heard from me by monday.

takachinss1001jp, 2020-12-12:
Thank you your Promt reply.
I'm sorry to have taken the trouble in a busy place.
I know ScratchMonky software is nice to work for HVPP,ISP,TPI with Arduino Uno which I made with Breadboard. and Thank your release code and Usermanual on git, I Used conveniently to rescue for blicked AVR.
If It's possoble, please remind which D0-13 are assined PORTB, PORTC ,PORTD and PORTE and A0-5 are assined to Arduino Leonard(ATmega32U4) are assined to PORTF(Boundary assignment and no A6&A7 like as Arduino Uno).　Portmapping is quite complicated so that I will inform the web as follows; 
It's written in Japanese,  you may see the table for processor which how assined I/O registor mapping. Please see https://ht-deko.com/arduino/portregisters.html.
Anyhow I 'm looking forward to remember and check.

Kazuhiro Haitaka

microtherion, 2020-12-12:
So I refreshed my memory; here's what's going on:

1. The bitmasks are not literally copied to the data direction registers. Only the `1` bits are used. Any pin corresponding to a `0` bit is left in the direction it was before.

2. The control signals are split across two different ports. `CTRL0,4-7` on `PORTF` and `CTRL2,3` on `PORTD`. `CTRL1` is not actually used.

```
CTRL  01234567
PORTD   23
PORTF 0   4567
```

Note that the names of these pins on the board are very confusing. `PF7` is `A0` etc. So here's a table for clarity:

Ctrl | Port | Pin
----|-----|---
CTRL0 | PF0 | A5
CTRL2 | PD2 | RX
CTRL3 | PD3 | TX
CTRL4 | PF4 | A3
CTRL5 | PF5 | A2
CTRL6 | PF6 | A1
CTRL7 | PF7 | A0

Hope that helps!

takachinss1001jp, 2020-12-13:
Thank you very much your Commentary concerning CTRL bit.
I have clear understood now.  By the way I knew CTRL1 is not used as in SMoHWIF_Leonard.h
117 typedef SMoHWIF_Port_Dual<SMoHWIF_PORT_D, 0x0C, 0,
118                          SMoHWIF_PORT_F, 0xF1, 0>          SMoHWIF_HVPP_Control;

If  so,  SMoHWIF_Port_Slow in SMoIF_Port.h might be Cause of malfunction.
81 template <uint8_t pin> class SMoHWIF_Port_Slow

I'm going to find out it. It may takes some time.
Anyway I will report again,  regardless of success or failure.
Thank you anyway.
I wish you a happy Christmas
Kazuhiro Haitaka

takachinss1001jp, 2021-02-10:
Thank you and I'm very sorry to keep you waiting.
 Unfortunetely Not only HVPP but also HVSP TPI ISP with Arduino Micro.
 Strange thing are nothing happened Arduino Micro D10(RESET/HVRESET)
and D11(VCC) with my 100MHz Oscilloscorpe. Despite Each Arduino Micro
Pin works as code of LED blink Sketch on Arduino IDE.
[image]
[image]
 First of all, I suspected to something wrong VCC control circuit,
However attached as Photo Arduino Nano works almost fine
Except Long duration of response.
 As you know This kind of assemble is Unreliable and Possibly poor contact.
 So frankly Speaking, I'm not prefer this Sir;