microtherion/ScratchMonkey #13: Is it possible to do HVPP on ATMega32U4?

geocine, 2020-12-11:
I don't know what I am asking but I am asking anyway, I recently got my ATMega32U4 bricked so hoping that HVPP is a solution.

microtherion, 2020-12-11:
I'm afraid I haven't been doing any electronics for a couple of years, but judging from the 32u4 data sheet, yes, HVPP should be possible. I assume you've already tried serial programming without success?

geocine, 2020-12-13:
Yes correct, I suspect I have locked myself out of RESET so looking into this option. If its not too much which part should I look into the datasheet?

microtherion, 2020-12-13:
Sections 28.5 and 28.6 in the datasheet, "Parallel Programming". For the mapping of the "CTRL" pins, see https://github.com/microtherion/ScratchMonkey/issues/12#issuecomment-743849239

I did not publish pinouts for SMD parts, because the pins on these are often not accessible enough to bother trying to reprogram them. So what's your plan. Are you desoldering the MCU, or did you install it in a through-hole adapter?

geocine, 2020-12-14:
I will try to figure out how to do this, thanks for the pointers. The pins your mentioned are all broken out on my custom arduino

closed 2021-03-08 by geocine